istoreos/istoreos #2394: 在线升级到24.10.1后，动态DDNS服务报错

qjmyymq, 2025-06-03:
在线升级到24.10.1后，动态DDNS服务报错，日志显示：
Error loading shared library libssl.so.1.1: No such file or directory (needed by /usr/lib/libisc-9.18.19.so)
Error loading shared library libcrypto.so.1.1: No such file or directory (needed by /usr/lib/libisc-9.18.19.so)
Error loading shared library libssl.so.1.1: No such file or directory (needed by /usr/lib/libdns-9.18.19.so)
Error loading shared library libcrypto.so.1.1: No such file or directory (needed by /usr/lib/libdns-9.18.19.so)
Error loading shared library libssl.so.1.1: No such file or directory (needed by /usr/lib/libisccfg-9.18.19.so)
Error loading shared library libcrypto.so.1.1: No such file or directory (needed by /usr/lib/libisccfg-9.18.19.so)
Error loading shared library libssl.so.1.1: No such file or directory (needed by /usr/lib/libirs-9.18.19.so)
Error loading shared library libcrypto.so.1.1: No such file or directory (needed by /usr/lib/libirs-9.18.19.so)
Error loading shared library libssl.so.1.1: No such file or directory (needed by /usr/lib/libbind9-9.18.19.so)
Error loading shared library libcrypto.so.1.1: No such file or directory (needed by /usr/lib/libbind9-9.18.19.so)
Error loading shared library libssl.so.1.1: No such file or directory (needed by /usr/lib/libns-9.18.19.so)
Error loading shared library libcrypto.so.1.1: No such file or directory (needed by /usr/lib/libns-9.18.19.so)
Error relocating /usr/lib/libisc-9.18.19.so: EVP_DigestInit_ex: symbol not found
Error relocating /usr/lib/libisc-9.18.19.so: SSL_SESSION_free: symbol not found
Error relocating /usr/lib/libisc-9.18.19.so: X509_sign: symbol not found
Error relocating /usr/lib/libisc-9.18.19.so: EC_KEY_free: symbol not found
Error relocating /usr/lib/libisc-9.18.19.so: EVP_aes_192_ecb: symbol not found
Error relocating /usr/lib/libisc-9.18.19.so: TLS_client_method: symbol not found
Error relocating /usr/lib/libisc-9.18.19.so: SSL_CTX_set_cipher_list: symbol not found
Error relocating /usr/lib/libisc-9.18.19.so: ERR_get_error: symbol not found
Error relocating /usr/lib/libisc-9.18.19.so: SSL_CTX_free: symbol not found
Error relocating /usr/lib/libisc-9.18.19.so: EVP_MD_CTX_md: symbol not found
尝试修复系统软件后仍然报错。

Nelson-Wcg, 2025-06-16:
同样报错，我是使用git的时候

duzp, 2025-07-04:
EasePi R1同样出现了这个问题，等待修复

closed 2025-10-16 by jjm2473