istoreos/istoreos #2410: 24.10.1 2025052311无法设置核显直通

guiyinrenshi, 2025-06-06:
昨晚我按照https://doc.linkease.com/zh/guide/istore/software/pve-vfio.html的教程给我的windows11配置核显直通，但是 IOMMU没起来，所以有了这个issue，后附我的grub文件和istoreos设置：

[grub.txt](https://github.com/user-attachments/files/20622780/grub.txt)

[image]

jjm2473, 2025-06-06:
你怎么判断iommu没起来

guiyinrenshi, 2025-06-06:
> 你怎么判断iommu没起来

[image]
pve无法使用

guiyinrenshi, 2025-06-06:
> 你怎么判断iommu没起来

root@iStoreOS:~# dmesg|grep -i iommu
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=fe638594-e6b3-fa9f-3a56-4d4f9cfc0302 rootwait noinitrd intel_iommu=on iommu=pt intel_iommu=on iommu=pt pcie_acs_override=downstream,multifunction disable_vga=1 vfio_iommu_type1.allow_unsafe_interrupts=1 kvm.ignore_msrs=1 modprobe.blacklist=radeon,nouveau,nvidia,nvidiafb,nvidia-gpu,snd_hda_intel,snd_hda_codec_hdmi,i915
[    0.024199] Kernel command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=fe638594-e6b3-fa9f-3a56-4d4f9cfc0302 rootwait noinitrd intel_iommu=on iommu=pt intel_iommu=on iommu=pt pcie_acs_override=downstream,multifunction disable_vga=1 vfio_iommu_type1.allow_unsafe_interrupts=1 kvm.ignore_msrs=1 modprobe.blacklist=radeon,nouveau,nvidia,nvidiafb,nvidia-gpu,snd_hda_intel,snd_hda_codec_hdmi,i915
[    0.024245] DMAR: IOMMU enabled
[    0.024266] DMAR: IOMMU enabled
[    0.304290] iommu: Default domain type: Passthrough (set via kernel command line)
[    0.414398] AMD-Vi: AMD IOMMUv2 functionality not available on this system - This is not a bug.
正常启用之后pve就可以使用这个IOMMU了，但是实际无法使用，因此判断其未启动

guiyinrenshi, 2025-06-06:
> `[ 0.024245] DMAR: IOMMU enabled`就说明intel iommu已经启用

那我排查下pve吧，估计是pve有的地方有问题

guiyinrenshi, 2025-06-06:
> 试试不要添加自定义的内核参数，只勾选“启用iommu”

我手动在grub添加的代码要删除吗

guiyinrenshi, 2025-06-06:
> 而且你手动修改的grub可能在固件升级时也不会保留

好，我回头删除掉

guiyinrenshi, 2025-06-06:
> 而且你手动修改的grub可能在固件升级时也不会保留
好像不行，所有的视频都是按B站的流程走的，添加的时候pve还是识别不出来，用的默认参数
[image]

[image]

root@iStoreOS:~# vim /etc/modules.d/vfio-pci
root@iStoreOS:~# dmesg|grep -i iommu
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=fe638594-e6b3-fa9f-3a56-4d4f9cfc0302 rootwait noinitrd intel_iommu=on iommu=pt
[    0.024175] Kernel command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=fe638594-e6b3-fa9f-3a56-4d4f9cfc0302 rootwait noinitrd intel_iommu=on iommu=pt
[    0.024221] DMAR: IOMMU enabled
[    0.304009] iommu: Default domain type: Passthrough (set via kernel command line)
[    0.414304] AMD-Vi: AMD IOMMUv2 functionality not available on this system - This is not a bug.
root@iStoreOS:~# dmesg|grep -i vfio
[    7.088329] VFIO - User Level meta-driver version: 0.3

jjm2473, 2025-06-06:
22.03固件中有成功过吗？或者在原版PVE中成功过吗？IOMMU是需要主板和BIOS支持的。

执行这行命令 `for d in /sys/kernel/iommu_groups/*/devices/*; do n=${d#*/iommu_groups/*}; n=${n%%/*}; printf 'IOMMU group %s ' "$n"; echo "${d##*/}"; done` 看看输出什么

guiyinrenshi, 2025-06-06:
> 22.03固件中有成功过吗？或者在原版PVE中成功过吗？IOMMU是需要主板和BIOS支持的。
> 
> 执行这行命令 `for d in /sys/kernel/iommu_groups/*/devices/*; do n=${d#*/iommu_groups/*}; n=${n%%/*}; printf 'IOMMU group %s ' "$n"; echo "${d##*/}"; done` 看看输出什么

root@iStoreOS:~# for d in /sys/kernel/iommu_groups/*/devices/*; do n=${d#*/iommu
_groups/*}; n=${n%%/*}; printf 'IOMMU group %s ' "$n"; echo "${d##*/}"; done
IOMMU group * *

guiyinrenshi, 2025-06-07:
> 这里有挺多BIOS的相应位置 https://us.informatiweb.net/tutorials/it/bios/enable-iommu-or-vt-d-in-your-bios.html 
[image]

主板上的设置是开着的

guiyinrenshi, 2025-06-07:
> 这里有挺多BIOS的相应位置 https://us.informatiweb.net/tutorials/it/bios/enable-iommu-or-vt-d-in-your-bios.html 

[image]

CPU也是支持的

guiyinrenshi, 2025-06-09:
> 你的BIOS开的只是VT-x，不是VT-d，可能主板或者BIOS不支持，问问主板厂商。
> 
> 或者你就试试原版pve能不能支持。

算了，不支持就不支持吧，不折腾了，辛苦大佬了