istoreos/istoreos #2415: pve安装的istoreos，quickstart不显示cpu温度，用ssh 互信sensors可以获取宿主机温度，请问下，需要修改的是luci-fan.lua这个文件吗？

binfoo, 2025-06-07:


APPX-Studio, 2025-06-20:
[image]
我和你一样 PVE的

binfoo, 2025-06-20:
> istoreos提供了一个hook方式自定义温度。写一个脚本放在 `/sbin/cpuinfo`，赋予可执行权限，脚本内容示例：
> 
> #!/bin/sh
> 
> echo "CPU 120.0°C"
> 文件准备好以后，执行下`/etc/init.d/quickstart restart`就会立即使用这个脚本了。

脚本写好了，执行quickstart restart没反应，需要修改quickstart文件吗？
温度变量使用的temp

[image]

jjm2473, 2025-06-20:
> 脚本写好了，执行quickstart restart没反应，需要修改quickstart文件吗？

但你截图执行的是start啊，不是restart。

> admin/status/overview 这里面怎么加温度和coremark分数呢？

参考其他openwrt固件，istoreos又没改概览。

qq1318659588, 2025-08-04:
写一个脚本替换 /sbin/cpuinfo 
```
#!/bin/sh

cpu_temp=$(cut -c1-2 /sys/class/thermal/thermal_zone0/temp)

echo "CPU ${cpu_temp}°C"
```

[image]