istoreos/istoreos #2431: 升级24.10后unblockneteasemusic无法启用

ziyieqingfeng, 2025-06-12:
升级24.10后unblockneteasemusic无法启用，可能于nodejs有关。

ziyieqingfeng, 2025-06-13:
> [@ziyieqingfeng](https://github.com/ziyieqingfeng)请您提供ssh或者终端执行下面命令的结果
> 
> ```
> opkg list-installed node
> opkg list-installed libcares
> ```

root@iStoreOS:~# opkg list-installed node
node - v16.20.2-1
root@iStoreOS:~# opkg list-installed libcares
libcares - 1.19.1-1

bcseputetto, 2025-06-13:
@ziyieqingfeng 请你升级这两个软件包
```
opkg update
opkg install node libcares
```
升级后，看下插件能不能正常运行。

`node`和`libcares` 是`unblockneteasemusic` 的依赖
但`node`和`libcares`在22.03和24.10里都不是自带的软件包。如果你在22.03安装`unblockneteasemusic`
这两个依赖会被自动安装，然后你升级到了24.10，这些依赖不会被升级到24.10的上的版本。
这个可以说并非unblockneteasemusic独有的问题。所有在22.03安装插件后被安装的依赖都有可能出现该问题。
比如`shadowsocks-libev`的依赖`libsodium`，这个软件包也是很多插件的依赖软件包
它没有一开始就包含在固件里，22.03，24.10都没有一开始就包含，会从软件源里安装。
这种不包含在固件里，从软件源后装的软件包，升级固件后都不会带动升级，所以造成了升级固件某些插件无法运行。

Asunfly, 2025-07-01:
我升级完貌似还是不能用