istoreos/istoreos #2432: Tailscale 插件

sinoserendipity, 2025-06-12:
iStoreOS 版本 24.10.1 2025060614，istore 的 tailscale 插件用的是iptables， 会导致系统上存在旧版 iptables 规则， 不鼓励混合使用 iptables 和 nftables 规则，这可能会导致流量过滤不完整，能否采用 https://github.com/asvow/luci-app-tailscale 来取代，它能选择 iptables 和 nftable, 而且更能更完整！

weikeet, 2025-06-15:
咋样，解决了吗？

sinoserendipity, 2025-06-15:
> 咋样，解决了吗？

解决方案是不安装 istore 里的 tailscale，到 https://github.com/asvow/luci-app-tailscale  下载 ipk 安装

WONGWAI1017, 2025-06-16:
升级到24.10，软件商店里的tailscale无法开启exit-node，用命令开启后，重启又失效了。不知道是不是这个问题导致的

sinoserendipity, 2025-06-16:
> 升级到24.10，软件商店里的tailscale无法开启exit-node，用命令开启后，重启又失效了。不知道是不是这个问题导致的

不要使用软件商店里的tailscale，问题不少，https://github.com/asvow/luci-app-tailscale  去下载 ipk安装，里面有安装指导

weikeet, 2025-06-16:
> > 升级到24.10，软件商店里的tailscale无法开启exit-node，用命令开启后，重启又失效了。不知道是不是这个问题导致的
> 
> 不要使用软件商店里的tailscale，问题不少，https://github.com/asvow/luci-app-tailscale 去下载 ipk安装，里面有安装指导

我是使用脚本安装的，配置一些防火墙就好了。参考 https://github.com/istoreos/istoreos/issues/2265

jjm2473, 2025-06-19:
tailscale已经支持nftables了，忽略“系统上存在旧版 iptables 规则...”

closed 2025-06-19 by jjm2473