istoreos/istoreos #2433: 更新了2025060614之后，ShadowsocksR Plus+ 未运行

justformg, 2025-06-13:


bcseputetto, 2025-06-13:
首先我不推荐继续在24.10上使用`ssr-plus`，它不支持fw4防火墙
如果你非要用，要解决升级后某些依赖没有升级的问题。
你用的是什么节点？ss？vless？vmess？请你提供更详细的信息

closed 2025-06-14 by jjm2473

yyywwweee, 2025-12-06:
> 首先我不推荐继续在24.10上使用`ssr-plus`，它不支持fw4防火墙 如果你非要用，要解决升级后某些依赖没有升级的问题。 你用的是什么节点？ss？vless？vmess？请你提供更详细的信息 最好有截图，日志

SSR PLUS 用了14年了 功能简单 配置简单 稳定    passwall虽然功能多 策略多 在稳定性方面差很多

bcseputetto, 2025-12-06:
> > 首先我不推荐继续在24.10上使用`ssr-plus`，它不支持fw4防火墙 如果你非要用，要解决升级后某些依赖没有升级的问题。 你用的是什么节点？ss？vless？vmess？请你提供更详细的信息 最好有截图，日志
> 
> SSR PLUS 用了14年了 功能简单 配置简单 稳定 passwall虽然功能多 策略多 在稳定性方面差很多

passwall不用分流，用单节点就跟ssrplus一模一样，而且你说你用ssr-plus 14年稳定，我也一直用passwall，也没有毛病，你说passwall稳定差很多的依据是什么？单纯是功能多策略多，就可以被作为不稳定的依据？而且passwall本身只是作为核心启动的脚本程序，只是创建配置文件喂给核心，创建防火墙规则让流量导向核心，启动后它就不参与任何代理行为了，实际代理的是核心程序(xray/sing-box那些)，这点ssr-plus也是一样的。

meichushengdexiuxia, 2025-12-10:
> passwall不用分流，用单节点就跟ssrplus一模一样，而且你说你用ssr-plus [...]

我也感觉passwall不稳定，访问网站的时候经常性的特别卡，有国外网站却走了国内网站DNS的情况。导致这类网站访问起来巨卡无比。但是SSRP+默认设置就不需要纠结DNS的问题。

bcseputetto, 2025-12-10:
@yyywwweee @meichushengdexiuxia 已经在24.10提供了 ssr-plus的 .run包
最近ssr-plus 提供了 nft的支持