istoreos/istoreos #2443: Docker只显示“配置”无法正常启动

Stark-Chen, 2025-06-16:
在升级22.03.7 2025051615后Docker无法启动
出现
[image]
[image]
打开后自动关闭
[image]
求大佬帮忙看看
[image]

bcseputetto, 2025-06-17:
提供日志
```
logread |grep docker
```

Stark-Chen, 2025-06-17:
> @Stark-Chen 
> 把注册表镜像的 `docker.1panel.live`去掉，保存应用看下

还是不行
root@iStoreOS:~# logread |grep docker
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: Section @zone[1] (wan) cannot resolve device of network 'vap_lan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv4 filter table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv4 nat table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv4 mangle table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv4 raw table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv4 filter table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-DHCP-Renew'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-Ping'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-IGMP'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-IPSec-ESP'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ISAKMP'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'linkease'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'wan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'wan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'lan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'docker'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv4 nat table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * NAT 'DockerNAT'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: wan will enable FULLCONE-NAT
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv4 mangle table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv4 raw table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:      - Using automatic conntrack helper attachment
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv6 filter table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv6 nat table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Clearing IPv6 mangle table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv6 filter table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-DHCPv6'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-MLD'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ICMPv6-Input'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ICMPv6-Forward'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-IPSec-ESP'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'Allow-ISAKMP'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Rule 'linkease'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'wan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'wan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'docker' -> 'lan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Forward 'lan' -> 'docker'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'lan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'wan'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:    * Zone 'docker'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd:  * Populating IPv6 nat table
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_lan_rule'
Tue Jun 17 18:29:54 2025 daemon.notice procd: /etc/rc.d/S99dockerd: Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_lan_rule'
[... cut ...] PLUGINS.md#version-header"
Tue Jun 17 18:31:48 2025 daemon.err dockerd[24518]: time="2025-06-17T18:31:48.919588246+08:00" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Tue Jun 17 18:31:48 2025 daemon.err dockerd[24518]: time="2025-06-17T18:31:48.920060383+08:00" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Tue Jun 17 18:31:48 2025 daemon.err dockerd[24518]: time="202

bcseputetto, 2025-06-18:
@Stark-Chen 你试一下重装docker，只要你docker目录没有被删除，重装docker也不会影响容器。在系统，软件包里
相关软件包有
`luci-i18n-dockerman-zh-cn`
`luci-app-dockerman`
`luci-lib-docker`
`docker`
`dockerd`
`docker-compose`

wo616288, 2025-06-18:
大佬，我的情况也跟这个一样，根据你发的这些软件全部删除重新安装也不行 ，还有别的办法吗？或者说原因是什么呢？@bcseputetto

wo616288, 2025-06-18:
> 提供日志
> 
> ```
> logread |grep docker
> ```

@Stark-Chen 我查到的日志只有两条
Thu Jun 19 00:34:41 2025 user.notice dockerd-init: Bridge device 'docker0' already defined in network config
Thu Jun 19 00:34:46 2025 daemon.err dockerd[11339]: failed to start daemon: Error initializing network controller: invalid CIDR address: 192.168.100.2

然后我才发现在配置页面的桥接地址一栏忘记把掩码位/32配上，配上之后重启就好了

rayangl, 2025-06-19:
`/etc/init.d/dockerd status`看看状态是否运行

`/etc/init.d/dockerd start` 没运行就手动启用试试

可能因为某些原因无法正确创建 socket 文件，重新生成一下
`service dockerd stop`
`rm -rf /var/run/docker.sock`
`/etc/init.d/dockerd start`

实在不行，就先重新安装一下，删除etc/config/dockerd
`opkg remove dockerd docker`
`opkg install dockerd docker`
`service dockerd start`

jjm2473, 2025-06-19:
@Stark-Chen 不会配置就别配置，把所有注册表镜像都删掉。

前面人家让你删掉"docker.1panel.live"，你没看到后面还有新的错误吗？

>  failed to start daemon: invalid mirror: unsupported scheme "" in "docker.1ms.run"

"docker.1ms.run"也是错误的，那你觉得你哪个注册表镜像是正确的？全部删掉。

jjm2473, 2025-06-19:
不要随便安装更新dockerd，openwrt官方的docker的防火墙是很难用的，无法跟openwrt的防火墙配合，可能会将docker容器的服务开放到公网上去。如果不小心更新了docker，那就修复系统软件。

closed 2025-06-19 by jjm2473