istoreos/istoreos #2446: LUCI界面的DHCP开关有BUG

fengyaochen, 2025-06-17:
iStoreOS 24.10.1 2025060614
LAN-DHCP配置-忽略接口设置无效，无论启用还是不启用，etc/config文件夹里的dhcp文件内的option dhcpv4 '不会更改'

fengyaochen, 2025-06-17:
这是24.10.1 2025060614里的配置
config dhcp 'lan'
	option interface 'lan'
	option start '16'
	option limit '150'
	option leasetime '2h'
	option dhcpv4 'enable'

这是22.03.7 2025050912里的配置
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option ignore '1'

bcseputetto, 2025-06-17:
@fengyaochen 你是不是跑去升级过软件包？

[image]

我这里测试通过luci 设置“LAN DHCP忽略此接口”，保存。先不要应用，点开右上角未保存的配置。
就可以看到上图，它就是执行了一个 `uci set dhcp.lan.ignore='1'`
体现在配置文件上 就是`option ignore '1' ` 而不是 `option dhcpv4 'disable'`
这个 `option dhcpv4 'disable'` 感觉像是别的插件往配置里面加的

fengyaochen, 2025-06-17:
> [@fengyaochen](https://github.com/fengyaochen) 你是不是跑去升级过软件包？
> 
> [image]
> 
> 我这里测试通过luci 设置“LAN DHCP忽略此接口”，保存。先不要应用，点开右上角未保存的配置。 就可以看到上图，它就是执行了一个 `uci set dhcp.lan.ignore='1'` 体现在配置文件上 就是`option ignore '1' ` 而不是 `option dhcpv4 'disable'` 这个 `option dhcpv4 'disable'` 感觉像是别的插件往配置里面加的

option dhcpv4 'disable 这一条是我全新安装24.10以后就有的，没有更新过软件源，LUCI执行的命令还是老版本表达方式，你如果装22.03再保留配置升级上来就没问题

bcseputetto, 2025-06-17:
@fengyaochen 全新安装，不会有这条，难道你是单口设备？

fengyaochen, 2025-06-17:
> [@fengyaochen](https://github.com/fengyaochen) 全新安装，不会有这条，难道你是单口设备？

猜对了，是单口设备，原先是ESXI双口的，这两天重新搞成了单口

bcseputetto, 2025-06-17:
`option dhcpv4 'disable‘`
别的软件也是可以添加的，它同样能起到禁用DHCP的作用。
它和编辑LAN那里操作不一样，所以LAN没有办法去掉这条

bcseputetto, 2025-06-17:
@fengyaochen 单口设备的话，iStoreOS会默认将LAN作为DHCP客户端，而且肯定不会启用DHCP

bcseputetto, 2025-06-17:
那这条大概就是检测到单口设备，然后脚本自动修改LAN为DHCP客户端时，加入了禁用DHCP的配置，这个配置就是用的
`option dhcpv4 'disable‘`

bcseputetto, 2025-06-17:
@fengyaochen 单口设备，默认会把LAN修改成DHCP客户端，这是22.03就有的功能，又不是24.10才加的

bcseputetto, 2025-06-17:
@fengyaochen 而且单口设备，你本来就是用来做旁路由了(除非你要做单臂路由，单臂路由不是旁路由)。
旁路由你还开什么DHCP服务器？

fengyaochen, 2025-06-17:
> [@fengyaochen](https://github.com/fengyaochen) 而且单口设备，你本来就是用来做旁路由了(除非你要做单臂路由，单臂路由不是旁路由)。 旁路由你还开什么DHCP服务器？

突发奇想做的单臂路由，想试试IPV6单臂旁路由

closed 2025-06-17 by fengyaochen

AlanWang2333, 2025-06-17:
> [@fengyaochen](https://github.com/fengyaochen) 而且单口设备，你本来就是用来做旁路由了(除非你要做单臂路由，单臂路由不是旁路由)。 旁路由你还开什么DHCP服务器？

你还忽略了一个选项，如果有人插了一个USB 无线网卡，想让单网口设备 既能收信号又能发 热点呢，那岂不是 热点的DHCP没了，得手动指定，这不残废了吗