istoreos/istoreos #2449: 关于小米路由器如何升级降级问题

mymickey, 2025-06-17:
[image]

[image]

[image]

如图我用的istoreos是23.8.16,但是在手动安装第三方插件的时候都是失败,比如 Unexpected archive size.或提示成功但是实际却没成功

1:我有办法降级到22.0.3吗,是否直接下载官方安装包就可以手动在后台升降级?
2:或者可以升级到24
3:或者告知我如何解决如上问题感谢大家了

bcseputetto, 2025-06-17:
你固件刚刷好的时候，空间应该是足够安装openclash的，为什么没装上？
你重置一下固件应该就会回到刚刷时的状态。
然后一上来就用openclash的包进行安装。安装成功后，服务里没有？

bcseputetto, 2025-06-17:
@mymickey 系统，备份与升级，执行重置
第一次用OpenWrt吗？

bcseputetto, 2025-06-18:
@mymickey 如果你用OpenWrt官方的，应该是可以装的，把 .run包上传到路由器上，因为闪存比较小，建议放在
/tmp 里，/tmp 是内存的映射，有内存大小的一般
然后执行
sh /tmp/xxxxxxxx.run
就可以安装了
用这个仓库24.10的包
https://github.com/bcseputetto/Are-u-ok