istoreos/istoreos #2451: 24.10重启后防火墙会关闭wan的入站

ayellowcat, 2025-06-18:
重启后，防火墙会自动关闭wan的入站，每次都需要需要手动打开，该怎么查找原因呢？

[image]

bcseputetto, 2025-06-18:
如果你是要放行IPv6，又搞不清楚该怎么设置，可以参考下面链接里的内容
https://github.com/istoreos/istoreos/discussions/1281#discussioncomment-8215959

bcseputetto, 2025-06-18:
如果你只是想开放路由器本机的地址到公网

[image]

参考这张图设置防火墙通信规则，如果你想路由器本机的IPv4和IPv6都开放，你可以不在高级设置里面限制“仅IPv6”
其余设置参考图里就行了

ayellowcat, 2025-06-18:
> 如果你只是想开放路由器本机的地址到公网
> 
> [image]
> 
> 参考这张图设置防火墙通信规则，如果你想路由器本机的IPv4和IPv6都开放，你可以不在高级设置里面限制“仅IPv6” 其余设置参考图里就行了

明白了，感谢解答，lucky设置的端口，这里也要设置一遍