istoreos/istoreos #2452: 升级到最新固件24.10.1后，web界面打不开，ssh登录报错

allan577, 2025-06-18:
Error loading shared library libpcre.so.1: No such file or directory (needed by /bin/grep)
Error relocating /bin/grep: pcre_fullinfo: symbol not found
Error relocating /bin/grep: pcre_jit_stack_free: symbol not found
Error relocating /bin/grep: pcre_assign_jit_stack: symbol not found
Error relocating /bin/grep: pcre_compile: symbol not found
Error relocating /bin/grep: pcre_study: symbol not found
Error relocating /bin/grep: pcre_maketables: symbol not found
Error relocating /bin/grep: pcre_exec: symbol not found
Error relocating /bin/grep: pcre_jit_stack_alloc: symbol not found
Error loading shared library libpcre.so.1: No such file or directory (needed by /bin/grep)
Error relocating /bin/grep: pcre_fullinfo: symbol not found
Error relocating /bin/grep: pcre_jit_stack_free: symbol not found
Error relocating /bin/grep: pcre_assign_jit_stack: symbol not found
Error relocating /bin/grep: pcre_compile: symbol not found
Error relocating /bin/grep: pcre_study: symbol not found
Error relocating /bin/grep: pcre_maketables: symbol not found
Error relocating /bin/grep: pcre_exec: symbol not found
Error relocating /bin/grep: pcre_jit_stack_alloc: symbol not found
Error loading shared library libpcre.so.1: No such file or directory (needed by /bin/grep)
Error relocating /bin/grep: pcre_fullinfo: symbol not found
Error relocating /bin/grep: pcre_jit_stack_free: symbol not found
Error relocating /bin/grep: pcre_assign_jit_stack: symbol not found
Error relocating /bin/grep: pcre_compile: symbol not found
Error relocating /bin/grep: pcre_study: symbol not found
Error relocating /bin/grep: pcre_maketables: symbol not found
Error relocating /bin/grep: pcre_exec: symbol not found
Error relocating /bin/grep: pcre_jit_stack_alloc: symbol not found

bcseputetto, 2025-06-18:
你在 22.03的时候，安装了 grep？而不是用的 busybox自带的grep？
```
opkg remove libpcre --force-removal-of-dependent-packages
opkg remove grep
```

allan577, 2025-06-19:
> 你在 22.03的时候，安装了 grep？而不是用的 busybox自带的grep？
> 
> ```
> opkg remove libpcre
> opkg remove grep
> ```
> 
> 24.10版本的 grep 依赖 libpcre2 旧版的 grep 依赖 libpcre，升级上来 libpcre不可用了

No packages removed.
看来只有重刷固件了

allan577, 2025-06-19:
官网最新固件无法下载