istoreos/istoreos #2453: iStoreOS 24.10 里的 iStore分身(iStoreDup) v1.1.6 似乎仍为旧版的  22.03.7 + iptables

ralopa673, 2025-06-19:
istore分身 生成的仍为  22.03.7 2024102509 旧版 + iptables

不知道，是否有考虑对此迭代呢？ 或者，如果同时提供 22.03 + iptables 和 24.10 + nftables 也挺好的。

zczyk, 2025-08-28:
分身确实不行，版本太低。导致很多软件都不能使用

kc-bao, 2025-10-08:
此问题已在 **酷友社** 上被提及。那里可能有相关详细信息：

https://www.koolcenter.com/t/topic/1561/1

jjm2473, 2025-11-07:
今天新上线了个插件 ShadoWRT ，也是分身，不过使用宿主的 /rom 作为基础镜像，所以不会有版本不一致问题。而且当宿主固件升级以后，分身也会跟着升级，保持版本一致，且数据全部保留。

jjm2473, 2025-11-07:
@zero617 快尝试istore里面的 ShadoWRT ，注意先停止以前的istoredup，以免冲突。

closed 2025-11-07 by jjm2473

zero617, 2025-11-08:
> 最近几天我们应该会在B站发布这个插件的视频教程，有兴趣的话可以关注下 “酷友社”

[image]软件包的存储空间似乎有些太小了，怎么扩大呢？

CirnoSalt, 2025-11-10:
> [@zero617](https://github.com/zero617) 快尝试istore里面的 ShadoWRT ，注意先停止以前的istoredup，以免冲突。

新版分身比旧版流畅好用太多了，旧版可以下架了