istoreos/istoreos #2463: 24.10版本uu加速器问题

MatrixxZENG, 2025-06-20:
升级到24.10版本后，uu加速器无法加速，始终显示加速失败，插件未联网。
降级到22.03后恢复正常。

因为降级后恢复了正常，所以也没抓到24.10的日志，麻烦大大帮忙排查一下，谢谢。

MatrixxZENG, 2025-06-25:
> > 可能是因为uu还没适配nftables
> 
> http://router.uu.163.com/api/plugin?type=openwrt-x86_64 更新一下吧 现在都9.2了 istore里面还是相当古早的版本

我用uu官网的shell脚本来更新发现24.10也不行，说运行完会显示SN，我自己跑SN显示不出来。

回滚22之后istore的能用我就没在意了

jjm2473, 2025-06-25:
昨天已经更新了啊

jjm2473, 2025-06-25:
还有问题就卸载重装

Eila-Shigure, 2025-06-25:
我的uu加速器升级之后可以加速，但是主机测试nat一直显示失败是什么情况？

Eila-Shigure, 2025-06-25:
> [@Eila-Shigure](https://github.com/Eila-Shigure) 你是从istore卸载重装的吗？如果不是就卸载重装

我是尝试过商店内卸载重装以及uu官网方式安装 结果是一样的 网络能连通 但是nat是失败

Mekiduan, 2025-06-25:
> > [@Eila-Shigure](https://github.com/Eila-Shigure) 你是从istore卸载重装的吗？如果不是就卸载重装
> 
> 我是尝试过商店内卸载重装以及uu官网方式安装 结果是一样的 网络能连通 但是nat是失败

可以考试试在【网络】 -> 【防火墙】 -> 【通信规则】中
添加三条规则
允许 tun163 的入站数据
允许来自 tun163 的转发数据
允许转发到 tun163 的数据

或者直接修改 /etc/config/firewall 文件，添加如下内容：
config rule
    option name 'Allow-uugamebooster-Input'
    option src '*'
    option target 'ACCEPT'
    option direction 'in'
    option device 'tun163'
    list proto 'all'

config rule
    option name 'Allow-uugamebooster-Forward-1'
    option src '*'
    option dest '*'
    option target 'ACCEPT'
    option direction 'in'
    option device 'tun163'
    list proto 'all'

config rule
    option name 'Allow-uugamebooster-Forward-2'
    option src '*'
    option dest '*'
    option target 'ACCEPT'
    option direction 'out'
    option device 'tun163'
    list proto 'all'