istoreos/istoreos #2468: 精简系统自带软件包

jibukeshi, 2025-06-21:
当前的 iStoreOS 系统固件自带了应用过滤、DDNS、易有云等软件包，这些软件包其实用不到，完全可以不内置在固件里面，而是有需要时通过 iStore 软件商店下载安装，精简系统体积。
另外，现在的本地文件管理严重依赖易有云软件包，其实我只需要本地文件管理功能而不需要远程访问，希望以后能独立出来。

Mekiduan, 2025-06-21:
卸载 OAF

opkg remove luci-app-oaf --force-removal-of-dependent-packages          # 卸载 OAF 的 LuCI 管理界面
opkg remove luci-i18n-oaf-zh-cn --force-removal-of-dependent-packages   # 卸载 OAF 中文语言包
opkg remove appfilter --force-removal-of-dependent-packages             # 卸载 appfilter 主程序
opkg remove kmod-oaf --force-removal-of-dependent-packages              # 卸载内核模块 kmod-oaf
rm -f /etc/config/appfilter                                             # 删除 OAF 的配置文件

卸载 DDNSTO
opkg remove ddnsto --force-removal-of-dependent-packages                # 卸载 ddnsto 主程序
opkg remove luci-app-ddnsto --force-removal-of-dependent-packages       # 卸载 ddnsto 的 LuCI 管理界面
opkg remove luci-i18n-ddnsto-zh-cn --force-removal-of-dependent-packages# 卸载 ddnsto 中文语言包
rm -f /etc/config/ddnsto                                                # 删除 DDNSTO 配置文件

卸载 易有云
opkg remove linkease --force-removal-of-dependent-packages              # 卸载 LinkEase 核心程序
opkg remove luci-app-linkease --force-removal-of-dependent-packages     # 卸载 LuCI Web 管理界面
opkg remove luci-i18n-linkease-zh-cn --force-removal-of-dependent-packages # 卸载 LinkEase 中文语言包
rm -f /etc/config/linkease                                              # 删除 LinkEase 配置文件

jibukeshi, 2025-06-23:
我知道可以删，但是由于 OpenWRT Overlayfs 的特性实际上没删干净，在 /rom 分区下仍然有。
我的意思是系统可以不自带这些，可以做到更精简。

jibukeshi, 2025-06-25:
@jjm2473 感谢解答，另外 1Panel 出 V2 版本了，iStore 里面的容器版能更新下吗🙏🏻