istoreos/istoreos #2495: 点击docker容器项出错

NC-0425, 2025-06-30:
点击docker容器菜单出现以下状况：

Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/model/cbi/dockerman/containers.lua:80: attempt to index field '?' (a nil value)

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:803:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:777:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:804:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1017:48)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^

请问怎么能够修复。谢谢！

jjm2473, 2025-07-05:
试试修复系统软件 https://github.com/istoreos/istoreos/wiki/FAQ

jjm2473, 2025-07-29:
补丁合并过来了，下次发固件会修复

closed 2025-08-08 by jjm2473