istoreos/istoreos #2511: tailscale低速问题

99477204, 2025-07-08:
### Describe the bug

以前一直挺快的跑满带宽，现在1、2百kb速度。不知道和istore版本有没有关系，从22直升到24的。在ssh界面输入tailscale配置信息后提示
Warning: UDP GRO forwarding is suboptimally configured on pppoe-wan, UDP forwarding throughput capability will increase with a configuration change.
See https://tailscale.com/s/ethtool-config-udp-gro

### OpenWrt version

iStoreOS 24.10.2 2025070410

### OpenWrt release

iStoreOS 24.10.2 2025070410

### OpenWrt target/subtarget

iStoreOS 24.10.2 2025070410

### Device

x86

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

a447948814, 2025-07-08:
没有关系吧，我这依然可以贷款跑满。

ralopa673, 2025-07-09:
没遇到你说的问题。你是不是两台tailscale之间是不同的宽带运营商，被跨网QoS了？

jjm2473, 2025-08-06:
如果是x86，那你可以用ventoy安装多个版本的istoreos测试，看看是否跟版本有关。