istoreos/istoreos #2515: 编译isotreos后没有经典的首页UI

karlhu85, 2025-07-08:
[image]
试了很多方法，都是没有，不知道是什么原因，手动安装is-opkg install luci-i18n-quickstart-zh-cn
是可以安装上首页UI的

我想请问一下这个首页UI不是开源的吗？我在代码中也找不到。

求大神指点应该如何编译才会有这个默认的首页？

bcseputetto, 2025-07-09:
编译的时候，源加了吗？
https://github.com/linkease/nas-packages-luci
这个源

jjm2473, 2025-07-11:
istoreos发布时都提供了编译配置文件，例如x86_64的两个文件：
* https://fw0.koolcenter.com/iStoreOS/x86_64_efi/feeds.conf
* https://fw0.koolcenter.com/iStoreOS/x86_64_efi/config.seed

使用这两个文件就能编译出一样的固件，除了某些软件可能有更新。

closed 2025-07-11 by jjm2473

karlhu85, 2025-07-15:
我找了配置，编译不正常报错😂😂😂，从梦想到梦想成真，皆可能我们用科技铸建桥梁，为您联接希望，达成梦想，成就无尽可能鱼鳍科技（上海）有限公司    ‪400-001-1020‬主营业务：5G带宽接入|无线网络覆盖|工业5G CPE|直播网络展会活动网络搭建|物联网卡流量|5g设备管理平台|直播设备定制-------- 原始邮件 --------发件人： Liangbin Lian ***@***.***>日期： 2025年7月15日周二 14:06收件人： istoreos/istoreos ***@***.***>抄送： karl ***@***.***>, Author ***@***.***>主    题： Re: [istoreos/istoreos] 编译isotreos后没有经典的首页UI (Issue #2515)jjm2473 left a comment (istoreos/istoreos#2515)
一样，固件从哪个文件夹下载的，就在哪找编译配置。