istoreos/istoreos #2518: 0704 版MWAN3模块不能叠加了

nbagui, 2025-07-09:
### Describe the bug

模块可以运行，但是叠加失效

### OpenWrt version

24.10 0704

### OpenWrt release

24.10 0704

### OpenWrt target/subtarget

24.10 0704

### Device

h88k

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

jjm2473, 2025-07-11:
根据 https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#installation 

```
opkg install iptables-nft ip6tables-nft
```
就支持nftables了。

如果还有问题就反馈给openwrt官方，mwan3是openwrt维护的。

closed 2025-07-11 by jjm2473

yatotm, 2025-08-28:
很奇怪，升级前是正常的，升级后就不行了，iptables-nft ip6tables-nft 这俩默认就带的，同样是升级版本后出现，从0718升级0801