istoreos/istoreos #2520: 外网测试异常，千兆宽带跑不满

King-hsw, 2025-07-10:
[image]

[image]

系统便捷工具-外网测速
这里面的节点是不是和网页的不一样啊
我用插件测速慢的那叫一个离谱

[image]

我电脑打开网页都500多M

[image]

不过还是有问题，我是千兆的宽带现在怎么也跑不满了

King-hsw, 2025-07-10:
好吧，换了一个节点主路由的速度正常了，要测台湾的才快，也是哭死了，内网的设备还是只有100Mbps多，都搞不懂是测速网的问题，还是哪里的问题了

jjm2473, 2025-07-11:
测速时别翻墙啊。

closed 2025-07-11 by jjm2473

jjm2473, 2025-07-15:
@King-hsw 那就是节点问题。

内网的话，istore有个内网测速，先测试内网是不是正常。如果内网只有100M，那就看看各个设备网卡是不是协商到100M了