istoreos/istoreos #2526: 可以增加一个类似mergerfs功能的插件吗？

albicoccca, 2025-07-12:
目前是用istoreos当软路由，同时接了三块硬盘再装点容器当nas用的，nas部分功能的需求比较轻量。软路由接少量硬盘兼顾一下存储、下载和同步照片，这应该是最基础也实用的功能了，毕竟够装istoreos的设备单独作为软路由有点浪费，类似用法的用户我不知道多不多，但肯定有。
在轻量需求下raid对于很多人并不必要，反而是一种负担，重要数据使用rsync定时同步到多个盘就足够了（大佬能不能也做个rsync插件啊。。。），raid增加了门槛和处理问题的难度，但多个挂载点确实也增加了管理复杂性，这种情况mergerfs就很清秀了，istoreos应该有向nas方面兼顾的意思，mergerfs相对来说足够灵活实用，至少是多了一个选择，希望大佬能考虑一下。

jjm2473, 2025-07-12:
你说的是这个吗？ https://github.com/trapexit/mergerfs

首先 mergerfs 是把多个挂载点合并成一个挂载点，也就是本来就要先挂载硬盘到特定文件夹，再组合这些文件夹。其次 mergerfs 是基于 FUSE 的，性能估计比较差。

现在istoreos不是会把硬盘都挂载到/mnt吗，把/mnt当成大硬盘用不就行了。

如果你只是想把多个文件夹组合，而且只读的话，还有一个性能更好的方式，就是mount --bind，把一个文件夹挂载到另一个文件夹上。例如我有两个硬盘分区分别挂载到了/mnt/ssd和/mnt/usb，其中/mnt/ssd/media/movie和/mnt/usb/media/movie都放着电影，我想组合放在一个文件夹/mnt/merged/movies里，那么可以创建/mnt/merged/movies/ssd和/mnt/merged/movies/usb，然后执行`mount --bind /mnt/ssd/media/movie /mnt/merged/movies/ssd; mount --bind /mnt/usb/media/movie /mnt/merged/movies/usb`。这样的话 /mnt/merged/movies 就包含了所有电影了，而且性能跟原生一样，要写的话也可以，只要写到里面的ssd或者usb文件夹就行。

albicoccca, 2025-07-12:
> 你说的是这个吗？ https://github.com/trapexit/mergerfs
>
> 首先 mergerfs 是把多个挂载点合并成一个挂载点，也就是本来就要先挂载硬盘到特定文件夹，再组合这些文件夹。其次 mergerfs 是基于 FUSE 的，性能估计比较差。
>
> 现在istoreos不是会把硬盘都挂载到/mnt吗，把/mnt当成大硬盘用不就行了。 [...]

mergerfs不是简单把多个挂载点合并的东西，应该算是虚拟成一块大硬盘，跟合并成一个挂载点有非常大不同，而且这个也不是最主要的目的，它的效果肉眼来看更像没有性能加成的raid1，太专业的名词我说不好，大概效果有这些；
1、假如有abc三块硬盘，将他们合并后，不仅仅是变成一个挂载点，他们空间是统一管理，自动平衡的，比如三个盘分别剩余1g，1g，2g的空间，需要存放一个3g的文件夹，他会将文件夹中的文件分别存于三块硬盘的这个目录下，确保你能放下（当然，如果是单个文件超了它确实不会切割存放），这个是mount --bind做不到的，也就是不用考虑到底写到/mnt里面哪个文件夹的问题，合并后它现在就是一块大硬盘了，而不是简单的大文件夹包小文件夹，说到底根本不是一种东西；
2、性能方面实际测试就是原来每块硬盘的性能，并没有明显损失。

albicoccca, 2025-07-13:
> 你说的是这个吗？ https://github.com/trapexit/mergerfs
>
> 首先 mergerfs 是把多个挂载点合并成一个挂载点，也就是本来就要先挂载硬盘到特定文件夹，再组合这些文件夹。其次 mergerfs 是基于 FUSE 的，性能估计比较差。
>
> 现在istoreos不是会把硬盘都挂载到/mnt吗，把/mnt当成大硬盘用不就行了。 [...]

我知道它是什么东西，不是突发奇想，之前在omv系统上长期使用，omv4以前是联合文件系统，后来mergerfs替代了它，更好用一些。
mergerfs不是简单把多个挂载点合并的东西，应该算是虚拟成一块大硬盘，跟合并成一个挂载点有非常大不同，而且这个也不是最主要的目的，它的效果肉眼来看更像没有性能加成的raid1，举个直白的例子，unraid那种硬盘管理方式跟这个基本一回事，大概效果有这些；
1、假如有abc三块硬盘，将他们合并后，不仅仅是变成一个挂载点，他们空间是统一管理，自动平衡的，比如三个盘分别剩余1g，1g，2g的空间，需要存放一个3g的文件夹，他会将文件夹中的文件分别存于三块硬盘的这个目录下，确保你能放下（当然，如果是单个文件超了它确实不会切割存放），这个是mount --bind做不到的，也就是不用考虑到底写到/mnt里面哪个文件夹的问题，合并后它现在就是一块大硬盘了，而不是简单的大文件夹包小文件夹，说到底根本不是一种东西；
2、性能方面实际测试就是原来每块硬盘的性能，并没有明显损失。

jjm2473, 2025-07-18:
你以前使用的时候，一般有什么选项配置

albicoccca, 2025-07-22:
> 你以前使用的时候，一般有什么选项配置

我觉得omv的插件功能刚刚好，可以了解下。
功能也上面有7个按钮，依次是“创建”“编辑”“重启”“删除”“平衡池”“去重”“文档”
里面最核心的“创建”主要有下面这些选项：
1、名称：不用多说。
2、文件系统：这个是重点，之前我关注点有问题，描述错了，这里也说明一下，mergerfs并不是合并挂载点，它直接合并的是磁盘。安装系统后没有挂载磁盘，在mergerfs这里新建的时候可以直接勾选多块磁盘创建新磁盘，然后能对合并新建的磁盘进行挂载，所以mergerfs更像raid1，而不是路径套娃。被合并的磁盘还可以单独进行挂载。
3、共享文件夹：这一项是要联动omv其他功能的，跟挂载有关。
4、路径：这个也不用管，新建后自动会填上磁盘的dev路径。
5、创建策略：选项很多，跟空间使用方式有关，没深研究过。
6、最小剩余空间：下面难得有写说明“当一个文件系统上可用空间达到最小值时，文件将不写入到此文件系统，如果所有文件系统可用空间都达到最小值时，那么这个文件会写入失败。”。
7、选项：这里可以敲一些参数，一般默认的不需要，但是我遇到过qb较新版本下载到合并后的磁盘报错不能下载的问题，mergerfs项目页当时有专门的说明，这里改写后正常，还是有必要的。

后面两个比较有用的功能是“平衡池”“去重”，“平衡池”应该是按照一定规则将占用率高的磁盘文件移动到占用率低的，但说实话这两个功能我都没怎么用过，核心还是前面的。

jjm2473, 2025-07-22:
根据mergerfs文档，mergerfs肯定是合并挂载点，只是omv让你选择分区而已，最后调用mergefs前还是会将挂载点作为参数。

closed 2025-07-29 by jjm2473

diave971, 2025-08-09:
> 你说的是这个吗？ https://github.com/trapexit/mergerfs
> 
> 首先 mergerfs 是把多个挂载点合并成一个挂载点，也就是本来就要先挂载硬盘到特定文件夹，再组合这些文件夹。其次 mergerfs 是基于 FUSE 的，性能估计比较差。
> 
> 现在istoreos不是会把硬盘都挂载到/mnt吗，把/mnt当成大硬盘用不就行了。
> 
> 如果你只是想把多个文件夹组合，而且只读的话，还有一个性能更好的方式，就是mount --bind，把一个文件夹挂载到另一个文件夹上。例如我有两个硬盘分区分别挂载到了/mnt/ssd和/mnt/usb，其中/mnt/ssd/media/movie和/mnt/usb/media/movie都放着电影，我想组合放在一个文件夹/mnt/merged/movies里，那么可以创建/mnt/merged/movies/ssd和/mnt/merged/movies/usb，然后执行。这样的话 /mnt/merged/movies 就包含了所有电影了，而且性能跟原生一样，要写的话也可以，只要写到里面的ssd或者usb文件夹就行。`mount --bind /mnt/ssd/media/movie /mnt/merged/movies/ssd; mount --bind /mnt/usb/media/movie /mnt/merged/movies/usb`

用上了，好像性能确实挺差的，我是好几个小的硬盘，想把它合并起来作为一个大的下载盘，usb下组raid太不稳定了，不知道大佬还有没有更好的方案