istoreos/istoreos #2529: 用商店中安装的Tailscale登录自己既有账户，秒崩。

Thomas1802-ai, 2025-07-12:
### Describe the bug

我的友善R5C以前忘了对着哪个视频教程，成功地运行了Tailscale并正常使用，在近日重装24.10版（7月4日版）后，通过istoreos商店安装Tailscale后，启用，点击生成的链接并输入自己原本的Tailscale账户信息，秒崩。字面意义上的秒崩。istoreos系统崩了，路由器无法正常启动和运行，只能重装系统固件。连试了2次，均秒崩，不敢再试了。访问Tailscale登录自己的账号，发现增加了新设备的信息，表明新设备在Tailscale添加成功，崩的只是istoreos。

### OpenWrt version

istoreos-24.10.2-2025070410-r5s-squashfs.img.gz

### OpenWrt release

istoreos-24.10.2-2025070410-r5s-squashfs.img.gz

### OpenWrt target/subtarget

istoreos-24.10.2-2025070410-r5s-squashfs.img.gz

### Device

友善R5C

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Thomas1802-ai, 2025-07-13:
在ChatGPT的帮助下，我最终在docker里装上了Tailscale。方法是通过ChatGPT编写一个创建到root文件夹里的“start-tailscale.sh”文件，然后去终端运行这个脚本，就可以直接创建Tailscale的docker容器。而后，通过生成的地址链接，很轻松地加入到自己原本的Tailscale账户中。注意在上述脚本中直接生成“--advertise-exit-node --advertise-routes”的内容，则加入到自己的Tailscale账户后，可以很容易设置“Subnets“和”Exit Node”功能。

jjm2473, 2025-07-18:
秒崩是啥，后台打不开了吗？机器的指示灯啥状态

jjm2473, 2025-07-19:
tailscale可不是我们提供的，是openwrt提供的，我们只是做了个界面而已。而且升级24.10的时候，我们已经改过tailscale的防火墙规则，也测试过能组网。要不你还是试试恢复出厂设置，不要添加任何第三方软件源，然后只安装和配置tailscale，看看能不能正常用。

lixplor, 2025-07-28:
我的斐讯N1也是升级最新版本tailscale 0.0.6后，在登录之后秒崩，机器网络不通，断电再上电也不行。

closed 2025-08-06 by jjm2473