istoreos/istoreos #2530: 应用商店OpenList网盘

99477204, 2025-07-13:
安装应用商店OpenList网盘，后续手动升级官方最新版，打开istore商店的openlist会显示OpenList 未运行，但实际打开http://192.168.100.1:5244可以正常使用。不知道是不是istore商店的luci和官方的不适配

[image]

99477204, 2025-07-13:
> 安装应用商店OpenList网盘，后续手动升级最新版，打开istore商店的openlist会显示OpenList 未运行，但实际可以正常使用。不知道是不是istore商店的luci和官方的不适配
> <img alt="Image" width="1912" height="756" src="https://private-user-images.githubusercontent.com/28886079/465690164-6bb6fd2b-ca36-4cee-bcc9-37cde581767d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIzNzk1MDAsIm5iZiI6MTc1MjM3OTIwMCwicGF0aCI6Ii8yODg4NjA3OS80NjU2OTAxNjQtNmJiNmZkMmItY2EzNi00Y2VlLWJjYzktMzdjZGU1ODE3NjdkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzEzVDA0MDAwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUzZTc1NWE5YTNhMjE3ZjU5NWJiN2M0ZjM3MTRhZGY4YWNmZjZiZTA4ODFkYTI3ZTdhMjljOTIxMzY2MTkzYTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.pbUEL_hhy59W2Dx9Qv3kKrwzqfs1SQ35A3HWUF3QK6k">

用的这个https://github.com/sbwml/luci-app-openlist

emmsevenQ, 2025-07-18:
> > 用的这个https://github.com/sbwml/luci-app-openlist
> 
> [#2517](https://github.com/istoreos/istoreos/issues/2517) 这里已经说过， sbwml 的openlist 和上游的 openlist 配置文件不一样，会错误。
> 
> 不能手动只 升级/安装 sbwml 的openlist 4.0.8，要么全套用他的；要么只用 iStore 商店的，不要手动升级。

执行命令后，依旧无法打开配置界面，且重启后5244端口被关闭，无法正常进入openlist

myiqzero, 2025-07-21:
现在这个又改名了，luci-app-openlist2，以后怎么更新呢？

xiangfeidexiaohuo, 2025-07-24:
> 现在这个又改名了，luci-app-openlist2，以后怎么更新呢？

最新源码 编译出来的 已经可以共存了，要自己编译下，改下访问端口 就好，我试过了。

closed 2025-07-29 by jjm2473