istoreos/istoreos #2542: iStoreOS 24.10.2 2025071110   一直断网重启   大神看下什么问题

huichangjiong, 2025-07-16:
Wed Jul 16 16:03:55 2025 daemon.warn odhcpd[8590]: No default route present, overriding ra_lifetime to 0!
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 64:ff:0a:df:b3:f8
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.100.248 64:ff:0a:df:b3:f8
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.248 64:ff:0a:df:b3:f8
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.248 64:ff:0a:df:b3:f8
Wed Jul 16 16:03:57 2025 daemon.info pppd[20136]: Remote message: Authentication success,Welcome!
Wed Jul 16 16:03:57 2025 daemon.notice pppd[20136]: PAP authentication succeeded
Wed Jul 16 16:03:57 2025 daemon.notice pppd[20136]: peer from calling number 1C:AE:CB:5C:06:64 authorized
Wed Jul 16 16:03:57 2025 daemon.notice pppd[20136]: local  LL address fe80::f48c:5642:16cd:8a84
Wed Jul 16 16:03:57 2025 daemon.notice pppd[20136]: remote LL address fe80::1eae:cbff:fe5c:0664
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: Joining mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::f48c:5642:16cd:8a84.
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: New relevant interface pppoe-wan.IPv6 for mDNS.
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: Registering new address record for fe80::f48c:5642:16cd:8a84 on pppoe-wan.*.
Wed Jul 16 16:03:57 2025 daemon.notice netifd: Network device 'pppoe-wan' link is up
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 100.103.15.79.
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: New relevant interface pppoe-wan.IPv4 for mDNS.
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: Registering new address record for 100.103.15.79 on pppoe-wan.IPv4.
Wed Jul 16 16:03:57 2025 daemon.notice ttyd[15920]: [2025/07/16 16:03:57:8335] N: rops_handle_POLLIN_netlink: DELADDR
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: Withdrawing address record for 100.103.15.79 on pppoe-wan.
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: Leaving mDNS multicast group on interface pppoe-wan.IPv4 with address 100.103.15.79.
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: Interface pppoe-wan.IPv4 no longer relevant for mDNS.
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 100.103.15.79.
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: New relevant interface pppoe-wan.IPv4 for mDNS.
Wed Jul 16 16:03:57 2025 daemon.info avahi-daemon[9298]: Registering new address record for 100.103.15.79 on pppoe-wan.IPv4.
Wed Jul 16 16:03:57 2025 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Wed Jul 16 16:03:57 2025 daemon.notice netifd: Interface 'wan' is now up
Wed Jul 16 16:03:57 2025 daemon.notice pppd[20136]: local  IP address 100.103.15.79
Wed Jul 16 16:03:57 2025 daemon.notice pppd[20136]: remote IP address 100.103.8.1
Wed Jul 16 16:03:57 2025 daemon.notice pppd[20136]: primary   DNS address 112.4.0.55
Wed Jul 16 16:03:57 2025 daemon.notice pppd[20136]: secondary DNS address 221.131.143.69
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) fc:7c:02:8e:b3:8f
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.100.165 fc:7c:02:8e:b3:8f
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: using nameserver 112.4.0.55#53
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: using nameserver 221.131.143.69#53
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Wed Jul 16 16:03:57 2025 daemon.warn odhcpd[8590]: No default route present, overriding ra_lifetime to 0!
Wed Jul 16 16:03:57 2025 daemon.notice netifd: Interface 'wan_6' is enabled
Wed Jul 16 16:03:57 2025 daemon.notice netifd: Interface 'wan_6' has link connectivity
Wed Jul 16 16:03:57 2025 daemon.notice netifd: Interface 'wan_6' is setting up now
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.165 fc:7c:02:8e:b3:8f
Wed Jul 16 16:03:57 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.165 fc:7c:02:8e:b3:8f android-b93f06457c5c9a11
Wed Jul 16 16:03:57 2025 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)

Ximilulieying, 2025-07-16:
你的pppoe-wan一直在波动，看wan口网线是不是没插稳

huichangjiong, 2025-07-19:
[image]

huichangjiong, 2025-07-19:
一分钟都坚持不了，DNS配置就转圈圈，显示检测中，再重连

bohaoist, 2025-07-19:
同样有问题