istoreos/istoreos #2552: 自己添加机型c8 668

qgz1, 2025-07-21:
通过编译发现刷了后没有无线哪一项，为什么呢

qgz1, 2025-07-22:
执行`iw list一下`，没有任何输出，但是驱动都打上了，这源码跟其它源码不一样吗？其它源码编译出来没缺过无线啊，操作都一样操作的，为什么呢

jjm2473, 2025-07-25:
其他源码，包括openwrt官方版本吗？

qgz1, 2025-07-26:
包括官方版本，打开驱动目录/rom/lib/firmware/mediatek和有无线一样，该有的都有但是用你们源码编译出固件刷上就没无线这一项

jjm2473, 2025-07-29:
iw list没有输出的话，肯定是没无线的，这是更底层的检查。首先，编译时选择跟openwrt同一个版本，例如24.10分支，然后，开机以后，看看系统日志有没有报错，以及执行lsmod查看内核模块有没有加载

qgz1, 2025-07-29:
https://github.com/Lienol/openwrt  内核有区别6.12  你们是6.6

a
***@***.***

        

         原始邮件

发件时间：2025年7月29日 09:13
收件人：istoreos/istoreos ***@***.***&gt;
抄送：qgz1 ***@***.***&gt;, Author ***@***.***&gt;

jjm2473, 2025-07-29:
编译openwrt 24.10分支就是6.6内核

qgz1, 2025-07-29:
用你们源码编译出来是6.6，Lienol源码编译出来就是6.12，好像都是24.10分支

主机名
OpenWrt

型号
NRadio C8-668GL

架构
ARMv8 Processor rev 4

目标平台
mediatek/filogic

固件版本
OpenWrt 24.10-SNAPSHOT r29588-c12c711bc6 / LuCI openwrt-24.10 branch 25.186.44806~9ceef01

内核版本
6.12.40

a
***@***.***

jjm2473, 2025-07-29:
你要对比也应该拿openwrt 24.10.2对比