istoreos/istoreos #2582: 上传速度被限制

zybsai, 2025-08-03:
网络中有一台群晖nas,以前从外网访问的时候上传速度可以达到正常上传带宽速度40Mb,但是最近上传速度只能有600多KB,当前istroeos版本为24.10,而且我也尝试了从A网络的电脑上网,B网络的nas中上传文件,感觉速度也只有6-700KB的样子,但是用手机中的测速软件在A网络中进行测试上传速度又是正常的40Mb左右

[image]
网络中使用设备上传数据时的上传速度

[image]
使用测速软件进行测试时的上传速度

不知道如何解决了

coolzhi, 2025-08-03:
不是istoreos的问题，网间限速了，精准策略，投诉都没用

怎么测试1用同一个ISP的手机卡走蜂窝看上行是否正常。2.凌晨1点后到早上9点前测试网间上行是否又正常了

zybsai, 2025-08-04:
> 不是istoreos的问题，网间限速了，精准策略，投诉都没用
> 
> 怎么测试1用同一个ISP的手机卡走蜂窝看上行是否正常。2.凌晨1点后到早上9点前测试网间上行是否又正常了

你的意思是运营商限制了上传速度吗?但是我用手机接入网络A,使用测速app,测出来上传速度是正常的,能达到第二张图片那样呢

zybsai, 2025-08-06:
> 应该是运营商给你限速了，最典型的就是上传限制6mb的带宽，而且测速延迟非常高。

那请问一下,如果被运营商限速了,那么为什么使用测速软件测出来上传速度会是正常的呢?

jjm2473, 2025-08-06:
是否跨运营商了？

coolzhi, 2025-08-06:
测速一般都是正常，不然你一投诉一个准啊。现在限速可以精准识别你这个流量是干嘛的。比如pt,pcdn这种的，所以为什么测速是正常实际使用不正常。

yangxiaoyang-me, 2025-08-12:
> > > 不是istoreos的问题，网间限速了，精准策略，投诉都没用
> > > 怎么测试1用同一个ISP的手机卡走蜂窝看上行是否正常。2.凌晨1点后到早上9点前测试网间上行是否又正常了
> > 
> > 
> > 我找了运营商，给我解限制了，哈哈哈
> 
> 那看来真的是和运营商有关啊,你打电话直接问他是不是给你限速了,大就直接告诉你是被限速了吗?不会忽悠你一下吗?看来我也要去投诉一下了.

直接投诉，有一口咬定运营商的问题，让他给你结果。其实客服可能会糊弄人，但是上门的小哥一般不会，直接和他沟通就行