istoreos/istoreos #2587: 关于IPV4公网和下载速度的问题

0512day1, 2025-08-05:
前置：我是J4125+i225b3的四口小主机物理机跑的istoreos，坐标江苏省苏州市，电信千兆

7月2号凌晨的时候，家里突然断网，遂把猫和软路由都重启了一下，然后发现我的V4公网被收回了，ip变成了100开头的大内网。
期间找电信客服投诉、反馈，告知后台显示我的这条宽带是“双栈公网”，但是我拨号只能得到大内网...折腾好久，迟迟未解决
这期间还出现了一个“网速波动很大”的问题。如下图

[image]

[image]
排除了下载工具和资源的问题（在另一个网络环境下，同一个下载工具同一个镜像资源同一时间下载，均能跑满带宽），我一直以为是我的小主机的软件或者硬件出了问题。尝试重刷immortal，什么插件都不安装，能稳定一下，但是不持久，也就三五分钟问题再现；换成硬路由拨号，也是同样。苦此久矣~~~

就在今天8月5日凌晨，突然发现istore拨号获取到的ip不是100开头了，变成了127开头的动态V4，遂立马尝试下载，果然和我预想的一样，下载速度恢复正常，主流资源能跑到八九十M/S。

由此，产生了一个疑问，来请教各位18cm：ipv4公网和下载速度是否有关？ 
欢迎大家友善讨论~

Ximilulieying, 2025-08-05:
ipv4公网提升的下载速度基本上提升的是p2p的下载速度，一般情况下载大型网站的资源就算拨号得到的是内网ip，基本上也都能跑满带宽的（多线程）。
还有，网络波动很大，排除掉内网原因，大概率是运营商那边出问题了。istoreos好像有内网测速，你可以试一下，看内网能不能跑满，如果不能，那原因基本上就在内网上了。

0512day1, 2025-08-05:
> ipv4公网提升的下载速度基本上提升的是p2p的下载速度，一般情况下载大型网站的资源就算拨号得到的是内网ip，基本上也都能跑满带宽的（多线程）。 还有，网络波动很大，排除掉内网原因，大概率是运营商那边出问题了。istoreos好像有内网测速，你可以试一下，看内网能不能跑满，如果不能，那原因基本上就在内网上了。

内网测速试过啦~2.5g的网卡基本上都是跑到2.3g，应该没啥问题。 
唉~  从有公网变成没公网，难受

0512day1, 2025-08-06:
> > 变成了127开头的动态V4
> 
> 应该是172开头的吧，127那就完蛋了。

打错了，是117开头的。半天就又给我收回了