istoreos/istoreos #2593: UPNP无法使用

bailuxianfeiying, 2025-08-06:
[image]

Lucky的stun内网穿透显示UPNP添加映射端口失败

[image]

比特彗星软件里的检测也显示失败

[image]

同时系统也不显示有其他转发的upnp

bailuxianfeiying, 2025-08-06:
系统日志

Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 48374, ...) FAILED
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 60834, ...) FAILED
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 54246, ...) FAILED
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:21 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 52422, ...) FAILED
Wed Aug  6 07:46:30 2025 daemon.notice miniupnpd[11268]: perform_stun: 1 response out of 4 received
Wed Aug  6 07:46:30 2025 daemon.notice miniupnpd[11268]: perform_stun: #0 external address or port changed
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 34670, ...) FAILED
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 49178, ...) FAILED
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 48466, ...) FAILED
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:30 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 56368, ...) FAILED
Wed Aug  6 07:46:39 2025 daemon.notice miniupnpd[11268]: perform_stun: 1 response out of 4 received
Wed Aug  6 07:46:39 2025 daemon.notice miniupnpd[11268]: perform_stun: #0 external address or port changed
Wed Aug  6 07:46:39 2025 daemon.warn miniupnpd[11268]: upnp_event_process_notify: connect(192.168.3.5:2869): Connection reset by peer
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: upnpevents_processfds: 0xaaaae18f0360, remove subscriber uuid:a89a05f3-f5b6-406f-ac5a-2a189eef05f2 after an ERROR cb: http://192.168.3.5:2869/upnp/eventing/oryhhakkdl
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 35206, ...) FAILED
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 58849, ...) FAILED
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 40086, ...) FAILED
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xaaaae18f0550, 6, 0) send_batch failed -4
Wed Aug  6 07:46:39 2025 daemon.err miniupnpd[11268]: perform_stun: add_filter_rule2(..., 42017, ...) FAILED
Wed Aug  6 07:46:48 2025 daemon.notice miniupnpd[11268]: perform_stun: 1 response out of 4 received
Wed Aug  6 07:46:48 2025 daemon.notice miniupnpd[11268]: perform_stun: #0 external address or port changed
Wed Aug  6 08:13:50 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 08:13:50 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xffff8ca5b290, 6, 2) send_batch failed -4
Wed Aug  6 08:13:53 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 08:13:53 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xffff8c9f8270, 6, 2) send_batch failed -4
Wed Aug  6 08:13:56 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 08:13:56 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xffff8c9b6590, 6, 2) send_batch failed -4
Wed Aug  6 08:13:59 2025 daemon.err miniupnpd[11268]: send_batch: mnl_cb_run returned -1
Wed Aug  6 08:13:59 2025 daemon.err miniupnpd[11268]: nft_send_rule(0xffff8c9d72b0, 6, 2) send_batch failed -4
[... cut ...]
Wed Aug  6 09:20:41 2025 daemon.err miniupnpd[31186]: nft_send_rule(0xffffbce91800, 6, 2) send_batch failed -4
Wed Aug  6 09:20:44 2025 daemon.notice miniupnpd[31186]: private/reserved address 100.78.59.93 is not suitable for external IP
Wed Aug  6 09:20:44 2025 daemon.notice miniupnpd[31186]: force_forwarding enable, private/reserved address 100.78.59.93 used as external IP
Wed Aug  6 09:20:44 2025 daemon.err miniupnpd[31186]: send_batch: mnl_cb_run returned -1
Wed Aug  6 09:20:44 2025 daemon.err miniupnpd[31186]: nft_send_rule(0xffffbceb2290, 6, 2) send_batch failed -4

bailuxianfeiying, 2025-08-06:
https://github.com/openwrt/luci/issues/7878

上游有提到因为luci的问题导致不显示，但是实际测试完全没法使用

jjm2473, 2025-08-06:
> Wed Aug 6 09:20:44 2025 daemon.err miniupnpd[31186]: send_batch: mnl_cb_run returned -1
Wed Aug 6 09:20:44 2025 daemon.err miniupnpd[31186]: nft_send_rule(0xffffbceb2290, 6, 2) send_batch failed -4

看你的日志中很多这样的，从未见过。要么你还是恢复出厂设置再测试吧。

bailuxianfeiying, 2025-08-06:
我是从22.03直接升级到24.10的，设备是H28K，我重刷再试试

jjm2473, 2025-08-06:
正常升级固件应该不会有问题，除非升级前你的系统有升级过一些系统软件包。

bailuxianfeiying, 2025-08-06:
重刷后，正常了