istoreos/istoreos #2602: istoreos24.10首页软件源配置更换失败

lavey404, 2025-08-08:
用分支istoreos-24.10编译出来的rootfs，做成docker镜像允许，发现首页的软件源配置更换失败
rootfs编译时间2025.08.05，我编译好的[下载地址](https://github.com/lavey404/Actions-iStoreOS-armsr-24.10/releases/download/2025.08.05-armv8/istoreos-armsr-armv8-generic-rootfs.tar.gz)

[image]

wukongdaily, 2025-08-09:
这个rootfs中原始的 `/etc/opkg/distfeeds.conf` 是怎样的 帖一下~

wukongdaily, 2025-10-08:
干脆自己手动替换算了 换成阿里云的

```
src/gz openwrt_core https://mirrors.aliyun.com/openwrt/releases/24.10.2/targets/armsr/armv8/packages
src/gz openwrt_base https://mirrors.aliyun.com/openwrt/releases/24.10.2/packages/aarch64_generic/base
src/gz openwrt_kmods https://mirrors.aliyun.com/openwrt/releases/24.10.2/targets/armsr/armv8/kmods/6.6.93-1-b388b781da4bc000e0f3f3a5a1a34eac
src/gz openwrt_luci https://mirrors.aliyun.com/openwrt/releases/24.10.2/packages/aarch64_generic/luci
src/gz openwrt_packages https://mirrors.aliyun.com/openwrt/releases/24.10.2/packages/aarch64_generic/packages
src/gz openwrt_routing https://mirrors.aliyun.com/openwrt/releases/24.10.2/packages/aarch64_generic/routing
src/gz openwrt_telephony https://mirrors.aliyun.com/openwrt/releases/24.10.2/packages/aarch64_generic/telephony
```