istoreos/istoreos #2605: 更新20250808后，接口反复重启

o879607, 2025-08-09:
多了一个接口，反复重启，应该是集成 map 和 ds-lite 以支持相应联网方式导致的，但是我这里用不到这种联网方式。

类型: 隧道接口
设备: dslite-wan_6_4
已连接: 否
接收: 0 B (0 Pkts.)
发送: 0 B (0 Pkts.)

dslite-wan_6_4

协议: 虚拟动态接口 (轻型双栈（RFC6333）)

Sat Aug  9 15:31:11 2025 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Aug  9 15:31:14 2025 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Aug  9 15:31:14 2025 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Aug  9 15:31:17 2025 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Aug  9 15:31:17 2025 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Aug  9 15:31:20 2025 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Aug  9 15:31:20 2025 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Aug  9 15:31:23 2025 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Aug  9 15:31:23 2025 daemon.notice netifd: Interface 'wan_6_4' is setting up now

jjm2473, 2025-08-11:
这不是你自己配置的吗？默认情况下没有这个接口，应该只有wan和wan6。你把你的/etc/config/network发出来看看。

jjm2473, 2025-08-21:
什么机型啊，都不回复怎么排查？

yujie787, 2025-08-22:
> > 我也是这样，删都删不了
> 
> 删掉这两个软件包就没了 ds-lite map

已经解决！

yujie787, 2025-08-22:
> 什么机型啊，都不回复怎么排查？

铭凡MS-01

jjm2473, 2025-08-22:
把/etc/config/network提取出来看看

jjm2473, 2025-08-22:
恢复出厂设置以后会不会出现

closed 2025-09-18 by jjm2473