istoreos/istoreos #2613: 访问AP/光猫插件开启后无法访问光猫

lu989898, 2025-08-12:
光猫地址192.168.1.1，openwrt能看到去光猫路由表，  lan口网段192.168.100.0/24，ssh到openwrt后台能ping通192.168.1.1光猫，但lan口下pc在192.168.100.0/24网段不能访问192.168.1.1光猫。

[image]

[image]

[image]

[image]

[image]

jjm2473, 2025-08-15:
你的电脑的网关是指向iStoreOS吗？看看电脑网关

jjm2473, 2025-08-15:
另外，电脑还有没有其他ip，尤其是包含192.168.1.1网段的ip？如果有就清除掉。

xfghvgnfyjssjgte, 2025-08-15:
这个自己新建虚拟网口，动态获得ip,默认网关关了，网关跃点设为100，防火墙划分在wan口不就行了，不需插件openwrt自己本身就支持的功能

lu989898, 2025-08-17:
> 另外，电脑还有没有其他ip，尤其是包含192.168.1.1网段的ip？如果有就清除掉。

只有192.168.100.0这个网段的一个ip

jjm2473, 2025-08-19:
电脑上执行`tracert 192.168.1.1`看看输出。另外，你有没有改过istoreos的防火墙？把istoreos的`/etc/config/firewall`提取出来看看

lu989898, 2025-08-19:
> 电脑上执行`tracert 192.168.1.1`看看输出。另外，你有没有改过istoreos的防火墙？把istoreos的`/etc/config/firewall`提取出来看看

防火墙只做了些端口映射仿佛没改啥其他的。

[image]

config defaults
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option fullcone '2'
	option synflood_protect '1'
	option drop_invalid '1'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option auto_helper '0'
	list network 'lan'

config zone
	option name 'wan'
	option output 'ACCEPT'
	option mtu_fix '1'
	option forward 'ACCEPT'
	option auto_helper '0'
	option input 'DROP'
	list network 'wan'
	list network 'wan6'
	list network 'vap_wan'
	list network 'vap_lan'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option target 'ACCEPT'
	list icmp_type 'echo-reply'
	list icmp_type 'echo-request'
	option family 'ipv4'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'
	option enabled '0'

config rule
	option name 'Support-UDP-Traceroute'
	option src 'wan'
	option dest_port '33434:33689'
	option proto 'udp'
	option family 'ipv4'
	option target 'REJECT'
	option enabled '0'

config include
	option path '/etc/firewall.user'

config include 'miniupnpd'
	option type 'script'
	option path '/usr/share/miniupnpd/firewall.include'
	option family 'any'
	option reload '1'

config nat 'docker_nat'
	option name 'DockerNAT'
	option proto 'all'
	option src 'lan'
	option target 'MASQUERADE'
	option extra '-i docker0'
	option family 'ipv4'
	option src_ip '172.16.0.0/12'

config rule 'linkease'
	option name 'linkease'
	option target 'ACCEPT'
	option src 'wan'
	option proto 'tcp'
	option dest_port '8897'
	option enabled '0'

config zone 'docker'
	option name 'docker'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option auto_helper '0'
	list subnet '172.16.0.0/12'

config forwarding 'docker_to_wan'
	option src 'docker'
	option dest 'wan'

config forwarding 'docker_to_lan'
	option src 'docker'
	option dest 'lan'

config forwarding 'lan_to_docker'
	option src 'lan'
	option dest 'docker'

config rule
	option name 'qb'
	option src 'wan'
	list src_ip '0.0.0.0'
	option target 'ACCEPT'
	option src_port '8090'
	option enabled '0'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'qb'
	option src 'wan'
	option src_dport '8090'
	option dest_ip '192.168.100.1'
	option dest_port '8080'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option src 'wan'
	option dest_ip '192.168.100.1'
	option src_dport '51280'
	option dest_port '51280'
	option name 'qb_Download'

config rule
	option name 'qb_download'
	option src 'wan'
	option target 'ACCEPT'
	list src_ip '0.0.0.0'
	option src_port '51280'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'tr'
	option src 'wan'
	option src_dport '61234'
	option dest_port '61234'
	option dest_ip '192.168.100.1'

config rule
	option name 'tr'
	option src 'wan'
	option src_port '61234'
	option target 'ACCEPT'
	list src_ip '0.0.0.0'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'tr_gui'
	option src 'wan'
	option dest_ip '192.168.100.1'
	option src_dport '9099'
	option dest_port '9099'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'bt_外网访问'
	option src 'wan'
	option src_dport '9091'
	option dest_ip '192.168.100.225'
	option dest_port '9091'

config include 'passwall'
	option type 'script'
	option path '/var/etc/passwall.include'
	option reload '1'

config include 'passwall_server'
	option type 'script'
	option path '/var/etc/passwall_server.include'
	option reload '1'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'xbox'
	option src 'wan'
	option dest_ip '192.168.100.111'
	option src_dport '3074'
	option dest_port '3074'

config include 'passwall2'
	option type 'script'
	option path '/var/etc/passwall2.include'
	option reload '1'

config include 'passwall2_server'
	option type 'script'
	option path '/var/etc/passwall2_server.include'
	option reload '1'

jjm2473, 2025-08-20:
@lu989898 我说的是电脑上执行`tracert 192.168.1.1`，要检查电脑访问192.168.1.1的路由是不是经过了192.168.100.1。

我看的防火墙有passwall的规则，如果你开了passwall，那就先关掉passwall以及任何翻墙/透明代理插件再测试。

closed 2025-08-20 by jjm2473

lu989898, 2025-08-22:
> [@lu989898](https://github.com/lu989898) 防火墙wan区域默认是打开了NAT，不知道你为什么关掉。不要说访问光猫了，我都怀疑你的电脑能不能正常上网。

再问问这个NAT应该如何设置？我的光猫桥接，openwrt拨号获取公网地址，这样还需要NAT地址翻译么？

[image]

[image]