istoreos/istoreos #2622: lan口隔一段时间就没网了

895090595, 2025-08-15:
lan口隔一段时间就没网了 , wan口是有网的通过ZeroTier能访问到
前两天升级了版本**iStoreOS 24.10.2 2025080813**
昨天断网2次,今天8点/13点又断网.每次重启之后就好了,不知道咋排查问题
**说明: 断网的时候没有做任何事情**
关键信息如下
[image]
[image]
[image]
[image]
[image]

895090595, 2025-08-15:
[image]

EvanGongka, 2025-08-22:
重新刷入 22.03.7 2025051615 版本  貌似解决了   我再持续更新