istoreos/istoreos #2625: transmission无法启动

DertahSama, 2025-08-16:
设备（我是刚刚新入手的，没用过其它的版本）：
> 主机名 | iStoreOS
> -- | --
> 型号 | FriendlyElec NanoPi R3S
> 架构 | ARMv8 Processor rev 0
> 目标平台 | rockchip/armv8
> 固件版本 | iStoreOS 24.10.2 2025080813 / LuCI openwrt-24.10 branch 25.220.17796~9a38f14
> 内核版本 | 6.6.93

在商店装了transmission，发现启动不了，报错如下：
```
Sat Aug 16 21:55:26 2025 daemon.err transmission-daemon[13146]: Error relocating /usr/bin/transmission-daemon: evthread_set_condition_callbacks: symbol not found
Sat Aug 16 21:55:26 2025 daemon.err transmission-daemon[13146]: Error relocating /usr/bin/transmission-daemon: evthread_set_id_callback: symbol not found
Sat Aug 16 21:55:26 2025 daemon.err transmission-daemon[13146]: Error relocating /usr/bin/transmission-daemon: evthread_set_lock_callbacks: symbol not found
Sat Aug 16 21:55:26 2025 daemon.info procd: Instance transmission::instance1 s in a crash loop 7 crashes, 0 seconds since last crash
```
重复卸载安装、重启设备也没有用。

经过我一顿摸索，发现这是依赖的问题，按照此处 https://github.com/transmission/transmission/issues/6387 的方法，装了一个新的依赖，再打个硬链接，就好了！
```
# opkg install libevent2-extra7
# rm -rf /usr/lib/libevent-2.1.so.7
# ln -s /usr/lib/libevent_extra-2.1.so.7.0.1 /usr/lib/libevent-2.1.so.7
```
希望官方下次更新能够修复这个问题！

w9315273, 2025-08-19:
> 经过我一顿摸索，发现这是依赖的问题，按照此处 [transmission/transmission#6387](https://github.com/transmission/transmission/issues/6387) 的方法，装了一个新的依赖，再打个硬链接，就好了！
> 
> ```
> # opkg install libevent2-extra7
> # rm -rf /usr/lib/libevent-2.1.so.7
> # ln -s /usr/lib/libevent_extra-2.1.so.7.0.1 /usr/lib/libevent-2.1.so.7
> ```

加了 `-s` 这是软链接

jjm2473, 2025-08-19:
如果是新机器，不然先试试恢复出厂设置再尝试安装

jjm2473, 2025-08-19:
libevent2最近一年都没更新，所以这个问题肯定是因为系统不干净，例如原本刷的22.03，安装过libevent2-extra，然后保留配置升级到24.10，要么就是使用了第三方的软件源，这可能导致安装的transmission不是官方版本

DertahSama, 2025-08-19:
> libevent2最近一年都没更新，所以这个问题肯定是因为系统不干净，例如原本刷的22.03，安装过libevent2-extra，然后保留配置升级到24.10，要么就是使用了第三方的软件源，这可能导致安装的transmission不是官方版本

清汤大老爷，我刚刚新买的R3S设备，处女线刷的官网最新的iStore，直接在iStore商店装的transimission啊！

jjm2473, 2025-08-19:
你看看这两个文件都是多久以前改的了，尤其libevent2已经一年多没变更了：
* https://github.com/openwrt/openwrt/blob/openwrt-24.10/package/libs/libevent2/Makefile
* https://github.com/openwrt/packages/blob/openwrt-24.10/net/transmission/Makefile

如果是全新的系统，怎么可能出现依赖问题。

你是不是使用了第三方软件源？要测试也很简单，恢复出厂设置，重新操作一次，然后把安装过程的日志发出来看看，之后测试transmission能否正常启动。

jjm2473, 2025-08-19:
不好意思，确实有依赖问题，不过软件包都是从openwrt仓库来的，不知道为何出现问题，而且libevent都是去年11月份编译的了。
https://downloads.openwrt.org/releases/packages-24.10/aarch64_generic/base/

[image]

jjm2473, 2025-08-19:
大概找到原因了，是因为固件中已经有libevent2，但是编译固件时没编译libevent2-pthreads（因为没用上），libevent2中就会关闭pthread相关功能：
https://github.com/openwrt/openwrt/blob/92d4f59709eac7422391187edbd85f2a14d5589b/package/libs/libevent2/Makefile#L125-L126

但是从线上安装的transmission又是在启用了pthread的情况下编译的，需要依赖开启了pthread的libevent2。

jjm2473, 2025-08-19:
这个坑提交给openwrt了 https://github.com/openwrt/openwrt/issues/19809 。

下个版本的istoreos会在编译时加上libevent2-pthreads（但不集成到固件中），这样就能支持transmission等官方软件包了。

jjm2473, 2025-08-22:
20250822 已修复

closed 2025-08-22 by jjm2473

jjm2473, 2025-08-28:
@nanshihui 设置成这个路径`/usr/share/transmission/public_html`