istoreos/istoreos #2626: 24.10.2 2025080813 USB3.0硬盘挂载后SMB读写速度问题

LakeXblue, 2025-08-16:
PC通过SMB协议向软路由挂载的USB硬盘写入文件速度可以达到100MB/S, 但是从软路由USB硬盘读取文件拷贝到PC却只有40MB/S, 百思不得其解。单独把USB硬盘直接插到PC上读写都大于100MB/s,说明硬盘本身没问题，是不是软路由固件问题呢？

w9315273, 2025-08-17:
看看网络拓扑

LakeXblue, 2025-08-17:
> 看看网络拓扑

主路由器是华硕AX86U，R1做旁路由。PC是WIN10系统，接在主路由器下面用SMB协议共享R1挂载的USB硬盘。诡异的是向硬盘里拷贝数据可以达到硬盘的满速100MB/s，读取却只有40MB/S,印象中之前22版本没这个问题，升级到24版本后这是第一次注意到这个问题。

haobinnan, 2025-08-19:
> > 对于非usb的 没啥效果。我觉得可以不用默认添加这个参数，按需添加，或者可以在ui上增加一个选项。
> 
> 网络共享本来就有这个选项 <img alt="Image" width="2000" height="1222" src="https://private-user-images.githubusercontent.com/3896703/479313340-bbf49bcb-f030-4ce7-a36c-b31f2fc2451d.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU1NzM4MDUsIm5iZiI6MTc1NTU3MzUwNSwicGF0aCI6Ii8zODk2NzAzLzQ3OTMxMzM0MC1iYmY0OWJjYi1mMDMwLTRjZTctYTM2Yy1iMzFmMmZjMjQ1MWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MTlUMDMxODI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWUzZGY3ZGU4YmVjNjBkNWQwMjkwYTRhOWE1YWI2ZmQwYmQ5MTlkMmYyNDQxYjFiMDMzZjljMTY2MjQzNDVhYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.bt1CS94A9v9N_DeHCDGAcmLN4zyJ-l39l3Gr2jJx1y0">

给开发维护人员攒点，考虑的非常周全！