istoreos/istoreos #2636: Port forwarding lost real IP端口转发丢失真实IP

TTcheng, 2025-08-21:
路由器端口映射到内网主机后，内网主机看到请求全部来自路由器，而不是真实IP
版本：24.10.2 2025080813

w9315273, 2025-08-21:
看看NAT规则

TTcheng, 2025-08-21:
> 看看NAT规则

[image]

jjm2473, 2025-08-22:
端口映射肯定会改IP，linux下都是如此

w9315273, 2025-08-25:
端口转发就是DNAT，只改写目的地，不会改源地址，只有SNAT和MASQUERADE才会改写源地址，所以只检查NAT规则就可以，软件卸载关掉看看，不知道用途不要硬开。

jjm2473, 2025-08-25:
哦，你这个DockerNAT规则是有问题的。
正常应该是这样：
[image]

jjm2473, 2025-08-25:
点一下编辑，将源地址改为“172.16.0.0/12”

TTcheng, 2025-08-25:
问题解决了，确实是DockerNAT配置的不对，感谢各位

closed 2025-08-25 by TTcheng

TTcheng, 2025-08-26:
> 正常情况下应该自动加这个ip段的，可能你很早之前就从22.03升级到24.10测试版，当时这个配置还没生效。

我应该是5月份升的24.10，当时还出现过luci界面布局异常，然后好像是重装通过备份恢复的。

jjm2473, 2025-08-26:
5月23之前的是测试版