istoreos/istoreos #2652: x86软路由，状态-概览菜单里的端口状态只显示一个网卡，其它3个不显示，但工作正常

000170, 2025-08-28:
如题

[image]

w9315273, 2025-08-28:
这是openwrt的bug

jjm2473, 2025-09-10:
有openwrt的issue链接吗

jjm2473, 2025-09-17:
@000170 @w9315273 你的网卡是不是在第一次开机前没接上？试试接上这些网卡，然后恢复出厂设置。

jjm2473, 2025-09-17:
@w9315273 出现问题以后，将系统里的`/etc/board.json`文件提取出来看看，然后执行`ls -l /sys/class/net/eth*/device/subsystem`把输出也发出来看看

jjm2473, 2025-09-17:
只要你的网卡都接上，恢复出厂设置/etc/board.json不可能只有eth0和eth1

jjm2473, 2025-09-17:
@w9315273 那你这个肯定不是istoreos，istoreos第一次开机会把所有eth*都添加到配置里。

jjm2473, 2025-09-17:
https://github.com/openwrt/openwrt/blob/main/package/base-files/files/etc/board.d/99-default_network
https://github.com/istoreos/istoreos/blob/istoreos-24.10/target/linux/x86/base-files/etc/board.d/99-default_network

w9315273, 2025-09-17:
我的是官方版, 但istoreos也有同样的问题, 等楼主试试看恢复出厂

jjm2473, 2025-09-18:
@w9315273 执行`ls -l /sys/class/net/eth*/device/subsystem`就知道了，肯定都是pci，前面我已经贴过openwrt的代码了，跟第一个网口一样的bus就会显示，跟网卡型号无关 https://github.com/openwrt/luci/blob/667871d68c11c05d952a702b4ddd1478dea64712/modules/luci-base/root/usr/share/rpcd/ucode/luci#L580-L621

你的虚拟机之所以不显示其他网卡，是因为eth0的bus是virtio，其他eth的bus都是pci。