istoreos/istoreos #2676: 重提lan的链路聚合问题，请问能否解决增加？

zsraylee, 2025-09-05:
重提lan的链路聚合问题，请问能否解决增加？
对上查找这个需求是22年的问题了。😭😭

zsraylee, 2025-09-08:
问题是ISTORE O里没有新建聚合设备的项

w9315273, 2025-09-08:
> 问题是ISTORE OS里没有新建聚合设备的项

安装 kmod-bonding

w9315273, 2025-09-08:
刚刚新安装的系统 你参考一下
[image]
[image]

zsraylee, 2025-09-08:
[image]
神经病了，一模一样，就是没有。。。。。。。

w9315273, 2025-09-08:
哦, 看了一下图, 你是在接口里找, 那不可能找得到的, bonding是设备不是接口, 说明你根本就没看
https://github.com/istoreos/istoreos/issues/2668#issuecomment-3252419720

zsraylee, 2025-09-08:
我也发现了，现在已建好，双千兆网口现在只能用bond 6，尝试802.3ad不行，确认交换机是支持的，已有一组聚合。但用2个网口聚合后有一个网口显示link down，只有一个网口link up。不知是什么原因，bond 6下正常。

zsraylee, 2025-09-08:
[image]

[image]
感谢回复，参考了上边你的手动创建参数，已经建立成功并正常使用。问题在高级设置中的“从设备选择哈希策略”，按你的项加了“layer3+4"就正常link up了。也可能是我的设备交换机配置要求是这样。感谢解答。

w9315273, 2025-09-10:
简单的讲 layer3+4 就是最好的