istoreos/istoreos #2677: 关于装机过程和运行过程中卡住的问题

jeffliut, 2025-09-06:
设备：研华j1900的x86物理机，网卡是i210

istoreos 在装机过程总是卡在“solarflare falocn driver unloading”，用rufus和PE写盘工具都试过了，偶尔能装上，重启后还是卡在这个位置。
已经反复装机几十次了，问题始终没有解决。从网上付费请了两个高手指导，也没装好。

现在装机装的我都麻了😮‍💨

jeffliut, 2025-09-06:
[image]
[image]
[image]

Zachwang007, 2025-09-07:
我也出现这个错误，也是x86主机，然后装的24.10版本的，怎么装都是到这里出问题，然后硬盘格式化重新装22.03版本就没问题。真是奇了怪了

jjm2473, 2025-09-10:
试试openwrt官方的24.10.2能不能启动。

另外，统计下出问题的cpu是什么？内存多少？有没有试过只接u盘不接硬盘启动？

jjm2473, 2025-09-17:
先反馈给openwrt官方吧，也许是内核版本问题，也许是内核配置问题。

然后你可以先用istoreos的22.03版本

jam7788520-spec, 2025-09-20:
我的是热启动就会卡在这里，关电源再重启就没事