istoreos/istoreos #2695: 希望增加rsync插件以及对mergerfs的2个建议

albicoccca, 2025-09-13:
抱歉，我又厚着脸皮来要新功能了
之前mergerfs的增加让我很惊喜，感谢大佬们的付出。

istoreos做轻nas的定位很符合大部分人日常需要，raid之类的数据备份机制对于普通用户来说可能是一种负担，但毕竟谁都需要一些备份手段，保障关键数据安全。
这里建议增加一个rsync的插件，rsync系统默认就带有，对于拥有两块及以上硬盘的人就可以很好满足数据备份功能，只要使用rsync将重要数据同步到不同硬盘除了不是实时之外跟raid1没有本质区别，只是要用命令行操作这点日常使用不太方便，也不符合istoreos的理念吧。
可否增加这样一个插件，拥有ui界面，可以自定义源目录、目标目录、基础参数等，能够查看日志，再加上设置定时任务功能，这样就可以很方便的备份重要数据，而且足够轻。
希望大佬们考虑一下。

另外对于mergerfs有两个建议：

1、插件ui中的设置选项“创建策略”默认为“现有路径，最多可用空间”，如果长期使用mergerfs的话建议改为“最多可用空间”，对于不了解机制的人来说，改为这个设置更合理。
“现有路径，最多可用空间”日常使用有个问题，它的优先级应该高于“最小可用空间”设置，举例：现在AB两块硬盘，合并为/mnt/disk，如果A中有目录music，B中没有，现在A已经没有剩余空间了，继续向/mnt/disk/music中存放文件，它会依据“现有路径”原则，继续尝试存放于A硬盘的/music，然后报错空间不足，而不会自动在B硬盘新建music目录并利用空间。
这种情况下手动在B中新建music可以解决问题，mergerfs默认设置成这样应该是强调用户自我管理，但是不了解的人可能会认为mergerfs没有发挥作用。
2、除了ui已有的创建策略、最小可用空间设置外，建议将以下内容填入“选项”，而不是用默认参数。
`defaults,allow_other,cache.files=partial,ignorepponrename=true,fsname=mergerfs_pool`
这些选项大部分时候有益于日常使用的，比如qb下载（使用libtorrent2.0时不确定，1.2还可以）和视频读取等，具体每项作用可以问ai，作为业余人解释不了太复杂的机制。

mergerfs这两处设置作为默认设置也许不太合适，可否考虑添加到ui界面说明中？这个工具现成中文资料确实很少，但对于轻nas比较有价值，没有直观的说明很多功能就浪费了。

xfghvgnfyjssjgte, 2025-09-30:
你要的sync插件，我很早就写出来了,开源，欢迎分享
https://github.com/xfghvgnfyjssjgte/Luci-app-Realsync

xfghvgnfyjssjgte, 2025-10-01:
调用的是系统自带命令，不必编译，无需打包，直接用安装脚本就可以用，看下使用说明即知如何食用

albicoccca, 2025-10-01:
大佬你这视频介绍里要是有个rsync我可能就搜到了。。。这个要是能合并到istoreos里就完美了> https://www.bilibili.com/video/BV1zmuyzhEci/?vd_source=325cccc5fbafbe131645618ba098f87e

albicoccca, 2025-10-01:
> https://www.bilibili.com/video/BV1zmuyzhEci/?vd_source=325cccc5fbafbe131645618ba098f87e

另外要是再加个计划任务就完美了